atmel-maxtouch/mxt-app #14: Display configuration checksums, NVRAM and in-memory

ndyer, 2015-06-09:


dangerousHobo, 2019-01-18:
Is there anyway currently to verify that a configuration file matches what is loaded in NVRAM?

mgong98, 2019-01-18:
Only way is to verify that the checksum of the file matches that of the device.

Thanks

Michael Gong
Sr Staff Applications Engineer
Microchip Technology
450 Holger Way
San Jose, CA 95134
C: (408) 316-2145
D: (408) 720-6501
E: Michael.Gong@microchip.com

dangerousHobo, 2019-01-18:
Thanks Michael, I can read the checksum for the .xcfg file but how can I read it from the device?

dangerousHobo, 2019-01-21:
Hi Michael,

That doesn't appear to work as expected. I have three configuration files and they all report a verified checksum, but the first one is the one loaded. (First and third one are the same file but different name/location)

```
# mxt-app --checksum /lib/firmware/maxtouch.xcfg
Version:1.28
Registered sysfs path:/sys/bus/i2c/drivers/atmel_mxt_ts/1-004a
Configuration read from /lib/firmware/maxtouch.xcfg in XCFG format
File checksum verified: 3519951

# mxt-app --checksum /root/Rev03.xcfg
Version:1.28
Registered sysfs path:/sys/bus/i2c/drivers/atmel_mxt_ts/1-004a
Configuration read from /root/Rev03.xcfg in XCFG format
File checksum verified: 7810530

# mxt-app --checksum /root/Rev04.xcfg
Version:1.28
Registered sysfs path:/sys/bus/i2c/drivers/atmel_mxt_ts/1-004a
Configuration read from /root/Rev04.xcfg in XCFG format
File checksum verified: 3519951
```

mgong98, 2019-01-21:
Can you send over your configs.  Let me double check.

Thanks

Sent from my Verizon, Samsung Galaxy smartphone

dangerousHobo, 2019-01-22:
Attached is a zip file containing the three files I mentioned above.
[tpk-touchscreen.zip](https://github.com/atmel-maxtouch/mxt-app/files/2783064/tpk-touchscreen.zip)

mgong98, 2019-01-22:
Hi Michael.  I have duplicated the error.  We will work on fixing this.

The –checksum parameter is supposed to print out verified if the file and chip content is the same.

If the content is not the same, the printout is supposed to be what is in the file and what is in the chip.

The calculated checksum from the chip is not working.

The verified printout is also printing in decimal format.

Thanks

Michael Gong
Sr Staff Applications Engineer
Microchip Technology
450 Holger Way
San Jose, CA 95134
C: (408) 316-2145
D: (408) 720-6501
E: Michael.Gong@microchip.com

mgong98, 2019-01-23:
Hi, Michael.  Sorry, I was mistaken regarding the –checksum parameter.

After reviewing the code, it appears the –checksum parameter is used to recalculate the checksum and compare it to the file checksum.

The calculated checksum is not done on the content of the chip, so the results you received were correct.

The only way to get the device checksum using mxt-app is to save a configuration and verify the value at the head of the .xcfg file (CHECKSUM=0xAABBCC).

We can add a parameter that reads from the chip directly on a later release of the tool.

Sorry for the confusion.

Let me know if you have questions.