atmel-maxtouch/mxt-app #16: Update configuration using extended format and verify it

ndyer, 2015-06-09:


dangerousHobo, 2019-01-18:
Is this already working in v1.28? Looks like I can update my config using the extended format.

```
# mxt-app --load /lib/firmware/maxtouch.xcfg
Version:1.28
Registered sysfs path:/sys/bus/i2c/drivers/atmel_mxt_ts/1-004a
Configuration read from /lib/firmware/maxtouch.xcfg in XCFG format
Writing config to chip
Configuration loaded
Backed up settings to the non-volatile memory
Configuration backed up
Sending reset command
Chip reset
```

mgong98, 2019-01-18:
Yes, the .xcfg extended format works.

By using either the .xcfg extension, the code will extract the configuration in extended format when saving.

Thanks

Michael Gong
Sr Staff Applications Engineer
Microchip Technology
450 Holger Way
San Jose, CA 95134
C: (408) 316-2145
D: (408) 720-6501
E: Michael.Gong@microchip.com