atmel-maxtouch/mxt-app #41: AUR Package for Arch Linux

razor-x, 2016-01-17:
Found this by way of https://github.com/raphael/linux-samus/issues/73. With the new kernel on the Pixel 2, we need to reset the touchpad on boot / resume.

Ideally I would like to install this as a package from the AUR (then add a systemd hook to deal with the touchpad issue).

Would you consider adding this to the [AUR](https://aur.archlinux.org/)?

ndyer, 2016-01-22:
Thanks for pointing the issue out to me. I do think it's probably a kernel driver bug (which is also my responsibility) and probably should be fixed there.

All the same, I will have a look at what's involved in adding the package. We haven't been involved in packaging this utility previously, but with the Pixel chromebooks we do now have some users who are installing generic Linux distributions and need mxt-app.

genkimarshall, 2016-10-22:
I made a package here: https://aur.archlinux.org/packages/mxt-app-git/. Hopefully I have done it correctly. @ndyer Please let me know if this is fine, or perhaps you would like to own the package or be added as a co-maintainer.

cowlicks, 2017-05-19:
The package worked for me, but I edited the config to remove the pandoc dependency since it added like 50mb. The package  is less than a mb installed without it.