atmel-maxtouch/mxt-app #77: I2C error when all tests, broken line, sensor variant invoked through script

ninadwalvekar2019, 2019-10-24:
I am calling all test, broken line , sensor variant test back to back in a script and this results in an i2c error. Attached script test_nodelay prints the error on the console. The error occurs after the first test. The second one fails
However if I put a delay as in the test.sh script everything works fine.

Details of the setup:- 
mxt-app version mxt-app 1.28-19-g4e9daa7
kernel version :- 4.14

Attached is a log file of the error:- mxt-app-error

Note that the commands are tried by adding a reset command after every test still the same behavior.

atmeltouchlab, 2019-10-25:
Can you reattach the log files and the script you are using.  They don't seem to be in the bug.

ninadwalvekar2019, 2019-11-07:
[logs.zip](https://github.com/atmel-maxtouch/mxt-app/files/3818793/logs.zip)

atmeltouchlab, 2020-01-20:
A delay has been added after the reset of device.  Modification added to https://github.com/atmel-maxtouch/mxt-app/commit/550d39855dc8df888df70183e272624e7d238a87.

atmeltouchlab, 2020-01-20:
Closing issue.

closed 2020-01-20 by atmeltouchlab