atmel-maxtouch/mxt-app #105: Info checksum error read=0005D6 with mxt640UD

biccius, 2022-11-28:
I'm running mxt-app from a raspberry pi using i2c native interface with a mxt640
It actually works on a mxt640U and i get following responses:

```
raspberrypi@raspberrypi:~/mxt-app $ ./mxt-app -d i2c-dev:1-004b -i

Device registered on i2c-dev adapter:1 address:0x4b

Family: 166 Variant: 1 Firmware V1.1.AA Objects: 41
Matrix size: X32Y20
Information Block CRC: 0xFE4DE3

Type Start Size Instances ReportIds Name
-----------------------------------------------------------------
T37   256   130     1        0-0    DEBUG_DIAGNOSTIC_T37
T44   386     1     1        0-0    SPT_MESSAGECOUNT_T44
T5    387    11     1        0-0    GEN_MESSAGEPROCESSOR_T5
T6    398     7     1        1-1    GEN_COMMANDPROCESSOR_T6
T68   405    73     1        2-2    SERIAL_DATA_COMMAND_T68
T38   478    64     1        0-0    SPT_USERDATA_T38
T71   542   200     1        0-0    SPT_DYNAMICCONFIGURATIONCONTAINER_T71
T110  742    40    12        0-0    SPT_SELFCAPTUNINGPARAMS_T110
T7   1222     7     1        0-0    GEN_POWERCONFIG_T7
T8   1229    15     1        0-0    GEN_ACQUISITIONCONFIG_T8
T15  1244    11     1        3-3    TOUCH_KEYARRAY_T15
T18  1255     2     1        0-0    SPT_COMMSCONFIG_T18
T19  1257    16     1        4-4    SPT_GPIOPWM_T19
T25  1273    16     1        5-5    SPT_SELFTEST_T25
T40  1289     7     1        0-0    PROCI_GRIPSUPPRESSION_T40
T42  1296    14     1        0-0    PROCI_TOUCHSUPPRESSION_T42
T43  1310    15     1        6-6    SPT_DIGITIZER_T43
T46  1325    18     1        7-7    SPT_CTECONFIG_T46
T47  1343    47     1        0-0    PROCI_STYLUS_T47
T56  1390    36     1        8-8    PROCI_SHIELDLESS_T56
T61  1426     5     6        9-14   SPT_TIMER_T61
T65  1456    23     3       15-17   PROCI_LENSBENDING_T65
T70  1525    10    20       18-37   SPT_DYNAMICCONFIGURATIONCONTROLLER_T70
T72  1725    89     1       38-38   PROCG_NOISESUPPRESSION_T72
T77  1814     2     1        0-0    SPT_CTESCANCONFIG_T77
T78  1816    12     1        0-0    PROCI_GLOVEDETECTION_T78
T79  1828     4     3        0-0    SPT_TOUCHEVENTTRIGGER_T79
T80  1840    14     1       39-39   PROCI_RETRANSMISSIONCOMPENSATION_T80
T81  1854    18     2       40-41   PROCI_UNLOCKGESTURE_T81
T93  1890    30     1       42-42   PROCI_TOUCHSEQUENCELOGGER_T93
T100 1920    68     1       43-60   TOUCH_MULTITOUCHSCREEN_T100
T104 1988    11     1        0-0    SPT_AUXTOUCHCONFIG_T104
T108 1999    75     1       61-61   PROCG_NOISESUPSELFCAP_T108
T109 2074     9     1       62-62   SPT_SELFCAPGLOBALCONFIG_T109
T111 2083    32     2        0-0    SPT_SELFCAPCONFIG_T111
T112 2147     5     1       63-63   PROCI_SELFCAPGRIPSUPPRESSION_T112
T113 2152     3     1        0-0    SPT_PROXMEASURECONFIG_T113
T115 2155    20     1       64-64   PROCI_SYMBOLGESTURE_T115
T116 2175   255     1        0-0    SPT_SYMBOLGESTURECONFIG_T116
T121 2430     3     1        0-0    PROCI_SENSOR_CORRECTION_T121
T132 2433    18     1       65-65   SPT_MESSAGEFILTER_T132
```

Connecting the same interface with a mxt640UD, i'm experiencing the following issue:

```
raspberrypi@raspberrypi:~/mxt-app $ ./mxt-app -d i2c-dev:1-004b -i

Device registered on i2c-dev adapter:1 address:0x4b
Info checksum error calc=04DFFC read=0005D6
raspberrypi@raspberrypi:~/mxt-app $ 
```

Do you have any tips i can try?

biccius, 2022-11-29:
Thank you for the reply

I think the problem is quite different.
We have a [MXT640UD-HA](https://www.microchip.com/en-us/product/ATMXT640UD) and we learned that the protocol of the 'UD-HA' version is completely different from that of the 'UD'.

By the way, I think your tool is incompatible with the 'UD-HA' version chip

Can you confirm this for me?

mgong98, 2022-11-29:
> EXTERNAL EMAIL: Do not click links or open attachments unless you know the content is safe

Yes, the protocol is different with the UD-HA chips.
They ae used for functionality safety and have the requirement for the TX Seq num and the CRC8 on address writes.

The mxt-app has been updated to talk to this device.
Please try the latest version of mxt-app and let me know if you have any issues.

I cannot tell from the screen dumps if you are using the latest.

biccius, 2022-12-01:
I'm currently using v.1.36
How do I enable replies with MXT640UD-HA ?

If i send the line command

`raspberrypi@raspberrypi:~/mxt-app $ ./mxt-app -d i2c-dev:1-004b -i`

I get wrong checksum calculation error

mgong98, 2022-12-01:
Please also try ./mxt-app -i

This uses the sysfs framework.
I will check on a board here.

biccius, 2022-12-02:
I currently have the raspberry's SDA and SCL pins connected directly to module as well as VCC and GND.

In this configuration the only way I have without recompiling a new kernel is to use the following command:

`/mxt-app $ ./mxt-app -d i2c-dev:1-004b -i`

which, however, fails for the HA version.

mgong98, 2022-12-02:
By default, both the non -d and -d option should work.
It is telling the mxt-app to use the sysfs interface versus the debugfs interface.

Do you have an option to recompile the mxt-app on Github.

https://github.com/atmel-maxtouch/mxt-app

mgong98, 2023-06-30:
I have just tested the 640UD and mxt-app with the following links.

https://github.com/atmel-maxtouch/maXTouch_linux.git

https://github.com/atmel-maxtouch/mxt-app.git

Both of these were from the master branch and working with i2c-dev and sysfs

Below is a dump from using mxt-app.  The 640UD-HA board uses firmware 1.0.AA.

Can you confirm that the Linux driver includes the new read and write routines for the HA device?
The latest updates to the Linux driver and mxt-app are required for HA devices.

Thanks

--------Using i2c-dev ---------------------

***@***.***:~# ./mxt-app -d i2- c-dev:1-004b

Device registered on i2c-dev adapter:1 address:0x4b

Device registered on debug_fs path:/sys/kernel/debug/atmel_mxt_ts/
Command line tool for Atmel maXTouch chips version: 1.36

Select one of the options:

Enter L:   (L)oad config file
Enter S:   (S)ave config file
Enter I:   Read (I)nfo block and Config CRC
Enter D:   Rea(D) individual object config
Enter W:   (W)rite individual object
Enter T:   Run sel(T)-test
Enter F:   (F)lash firmware to chip
Enter B:   (B)ackup the config data to NVM
Enter R:   (R)eset the maxtouch device
Enter C:   (C)alibrate the maxtouch device
Enter M:   Display raw (M)essages
Enter U:   D(U)mp Diagnostic data
Enter Q:   (Q)uit the application
i
Reading info block.....

Family: 166 Variant: 21 Firmware V1.0.AA Objects: 43
Matrix size: X32Y20
Information Block CRC: 0x4C9291

Type Start Size Instances ReportIds Name
-----------------------------------------------------------------
T117  268   215     6        0-0    SPT_DATACONTAINER_T117
T37  1558   130     1        0-0    DEBUG_DIAGNOSTIC_T37
T144 1688     2     1        0-0    SPT_MESSAGECOUNT_T144
T5   1690    12     1        0-0    GEN_MESSAGEPROCESSOR_T5
T6   1702     7     1        1-1    GEN_COMMANDPROCESSOR_T6
T38  1709    64     1        0-0    SPT_USERDATA_T38
T14  1773    65     1        0-0    PROCI_KEYTHRESHOLD_T14
T16  1838    33     1        0-0    PROCG_SIGNALFILTER_T16
T71  1871   200     1        0-0    SPT_DYNAMICCONFIGURATIONCONTAINER_T71
T110 2071    40    12        0-0    SPT_SELFCAPTUNINGPARAMS_T110
T118 2551     3     1        2-2    SPT_DATACONTAINERCTRL_T118
T141 2554     3    32        0-0    UNKNOWN_T141
T145 2650     1     1        0-0    SPT_IGNORENODESCONTROL_T145
T7   2651     7     1        0-0    GEN_POWERCONFIG_T7
T8   2658    15     1        0-0    GEN_ACQUISITIONCONFIG_T8
T10  2673    11     1        3-3    SPT_SELFTESTCONTROL_T10
T11  2684     4     1        0-0    SPT_SELFTESTPINFAULT_T11
T12  2688    10     2        0-0    SPT_SELFTESTSIGLIMIT_T12
T15  2708    11     1        4-4    TOUCH_KEYARRAY_T15
T18  2719    12     2        0-0    SPT_COMMSCONFIG_T18
T19  2743    16     1        5-5    SPT_GPIOPWM_T19
T24  2759    19     1        6-9    PROCI_ONETOUCHGESTUREPROCESSOR_T24
[... 90 lines cut ...]
On my /sys/bus/i2c/drivers folder i have "stmpe-i2c" that i think is the driver for my i2c device on raspberry pi3b+

With a i2c scan command i get, so i think my hardware should be ok

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

40: -- -- -- -- -- -- -- -- -- -- -- 4b -- -- -- --

50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --

70: -- -- -- -- -- -- -- --

But when i launch the command /mxt-app -i -d i2c-dev:1-004b i get the following error:

Info checksum error calc=6A194D read=0005D6

So it seems that the device answer to read info block request with a wrong checksum.

With the "U" version it worked perfectly and looking at the library it seemed to me that the "HA" part is actually considered, but I can't get it to work.

Please help.

biccius, 2023-07-06:
It doesn't seem to work and I honestly can't figure out why.

The kernel on my raspberry is 6.1.37 and /sys/bus/i2c/drivers/atmel_mxt_ts/ directory exist but folder "1-004B" is not present.

Only SDA, SCL and GND pins are connected between 640UD-HA and rpi

[image]

I've checked with oscilloscope and signals are ok

I continue to get checksum error calculation

mgong98, 2023-07-06:
If you do not have full folder down to the 1-004B, then the driver didn’t really load correctly.
The 1-004b is created when the OS finds the device on Bus 1 and slave address 0x4b.
0x4B being the only slave address we support for the chip.

The i2c-detect -y 1 scans all of bus 1 but doesn’t mean the driver was installed.

Can you double check your device tree and all the other files that are associated with the compile of the atmel_mxt_ts.c

KCONFIG, Makefile in the /drivers/input/touchscreen directory of your kernel.
.config at root for your kernel should have CONFIG_TOUCHSCREEN_ATMEL_MXT=y.

Please see the WIKI below.

https://github.com/atmel-maxtouch/maXTouch_linux/wiki

[immagine]<https://user-images.githubusercontent.com/27350014/251427200-2fa66190-303b-45f1-973f-ee125ffa51e5.png>

mgong98, 2023-07-06:
Hi, Fabrizio.
Can you double check where the /CHG line is connected?
This would be needed if accessing the part in general and could effect the bootup.

Thanks

biccius, 2023-07-07:
the CHG and RESET lines are kept high by pull-up as shown in the module diagrams:

[image]

When we use "pure" I2C interface with mxt-app, is required the control of CHG line via GPIO?

Thanks for the important support

mgong98, 2023-07-07:
Can you provide a direct email address for further assistance?

Thanks

[immagine]<https://user-images.githubusercontent.com/27350014/251663288-1b47b4f4-5ecf-43cf-9e61-acbce629ad09.png>

mgong98, 2023-07-07:
Hi, Fabrizio.

I believe I know the problem now.  Because the driver is not showing up, the kernel has not created the crc_enabled file both for the sysfs interface and the debugfs interface.

The file is created to indicate the maXTouch device supports the HA protocol.

For sysfs, the file is found in the /sys/bus/i2c/drivers/atmel_mxt_ts/1-004b/ directory.

For debugfs, the file is found in the /sys/kernel/debug/atmel_mxt_ts/ directory.

There is no way to tell what I2C protocol is supported when using a chip with the HA protocol, so this needs to be flagged by the Linux driver.
Along with this is the tx seq no byte that increments on each write.

The Tx seq no starts from 0x00 when the maXTouch device boots.  The value is stored in the directory to be passed to mxt-app.

If using mxt-app standalone without any driver, these flags are unknown, so the mxt-app fails and reads incorrect data.

In this case the Infoblock does not match.

I would first look at why the atmel_mxt_ts driver is not getting installed.

mgong98, 2023-07-07:
I can look into passing an argument to indicate the use of the HA protocol in the i2c-dev /debugfs interface and a mechanism to sync up the tx seq no#.

biccius, 2023-07-10:
Hi Michael,

thank you for the support.
Yes, i think you've hit the problem. 

Actually we're interested only in a workaround to makes the mxt-app works with a "HA" module and this is the reason i've opened this issue.

We've opened also a similar thread on the Microchip's forum and i think the problem is similar.
https://forum.microchip.com/s/topic/a5C3l000000BphrEAC/t391207

You can contact me here:
https://www.linkedin.com/in/fabrizio-benatti/

We ask you to release a beta branch that however makes the HA module usable with mxt-app

biccius, 2023-07-11:
From a clean installation of rpi i do following steps:

1. Install some packages required to building and compiling.
`sudo apt install raspberrypi-kernel-headers build-essential bc git wget bison flex libssl-dev make libncurses-dev`

2. Create a directory for us to work in.
`mkdir kernel
cd kernel`

3. Clone the latest kernel sources using GIT.
`sudo git clone --depth=1 https://github.com/raspberrypi/linux`

4. Open menuconfig
`sudo make menuconfig`

5. Compile the kernel, modules, and device tree blobs.
` sudo make -j4 zImage modules dtbs`

6. Install compiled modules
`sudo make modules_install`

7. Copy kernel, modules and other boot files
`sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
sudo cp arch/arm/boot/zImage /boot/kernel-maxtouch.img`

8. Configure kernel to use new image generate editing "/boot/config.txt"
`kernel=kernel-maxtouch.img`

9. Reboot
`sudo reboot`

In attachment the ".config" with my configuration

[config.txt](https://github.com/atmel-maxtouch/mxt-app/files/12013229/config.txt)

`CONFIG_TOUCHSCREEN_ATMEL_MXT=y
CONFIG_INPUT_TOUCHSCREEN=y`

After reboot my Linux version is :
`
raspberrypi@raspberrypi:~ $ uname -rms
Linux 6.1.37-gbb592ecdb99b armv7l
`

Directory "/sys/bus/i2c/drivers/atmel_mxt_ts" exist, but it's empty inside
[image]

Directory "/sys/kernel/debug/atmel_mxt_ts/ " doesn't exist
[image]

The "dmesg" command filtered for "i2c" says that:

`**raspberrypi@raspberrypi:~ $ dmesg | grep i2c**
[    0.049474] i2c-core: driver [dummy] registered
[    0.120065] i2c-core: driver [pca953x] registered
[    0.120114] i2c-core: driver [pcf857x] registered
[    0.125553] i2c-core: driver [max8973] registered
[    0.126343] i2c-core: driver [tps51632] registered
[    0.126375] i2c-core: driver [tps62360] registered
[    0.127655] i2c-core: driver [act8945a] registered
[    0.127692] i2c-core: driver [stmpe-i2c] registered
[    0.127723] i2c-core: driver [tps65217] registered
[    0.127753] i2c-core: driver [tps65910] registered
[    0.127783] i2c-core: driver [max8907] registered
[    0.127808] i2c-core: driver [max8997] registered
[    0.127836] i2c-core: driver [max8998] registered
[    0.127867] i2c-core: driver [tps6586x] registered
[    0.128031] i2c-core: driver [tps65090] registered
[... 33 lines cut ...]
[    2.200771] i2c-core: driver [sbs-battery] registered
[    2.201097] i2c-core: driver [lm90] registered
[    2.201132] i2c-core: driver [lm95245] registered
[    2.224994] i2c-core: driver [mpu3050-i2c] registered
[    2.225014] i2c-core: driver [isl29018] registered
[    2.225029] i2c-core: driver [isl29028] registered
[    2.225048] i2c-core: driver [ak8975] registered
[    2.246985] i2c_dev: adapter [bcm2835 (i2c@7e804000)] registered as minor 1
[    2.247080] i2c i2c-1: adapter [bcm2835 (i2c@7e804000)] registered
[    2.247093] i2c i2c-1: of_i2c: walking child nodes
[    2.247105] i2c i2c-1: This adapter dropped support for I2C classes and won't auto-dete                                                                    ct ics932s401 devices anymore. If you need it, check 'Documentation/i2c/instantiating-devi                                                                    ces.rst' for alternatives.
[    2.247119] i2c i2c-1: This adapter dropped support for I2C classes and won't auto-dete                                                                    ct lm90 devices anymore. If you need it, check 'Documentation/i2c/instantiating-devices.rs                                                                    t' for alternatives.
[    2.247127] i2c i2c-1: This adapter dropped support for I2C classes and won't auto-dete                                                                    ct lm95245 devices anymore. If you need it, check 'Documentation/i2c/instantiating-devices                                                                    .rst' for alternatives.
[    2.249142] i2c_dev: adapter [bcm2835 (i2c@7e805000)] registered as minor 2
[    2.249219] i2c i2c-2: adapter [bcm2835 (i2c@7e805000)] registered
[    2.249232] i2c i2c-2: of_i2c: walking child nodes
[    2.249243] i2c i2c-2: This adapter dropped support for I2C classes and won't auto-dete                                                                    ct ics932s401 devices anymore. If you need it, check 'Documentation/i2c/instantiating-devi                                                                    ces.rst' for alternatives.
[    2.249256] i2c i2c-2: This adapter dropped support for I2C classes and won't auto-dete                                                                    ct lm90 devices anymore. If you need it, check 'Documentation/i2c/instantiating-devices.rs                                                                    t' for alternatives.
[    2.249263] i2c i2c-2: This adapter dropped support for I2C classes and won't auto-dete                                                                    ct lm95245 devices anymore. If you need it, check 'Documentation/i2c/instantiating-devices                                                                    .rst' for alternatives.`

If i request the i2c scan using the "i2cdetect" command the module address is recognized

[image]

If you have other test to suggest i'm here available

mgong98, 2023-07-11:
The driver you are using isn’t compatible with our HA protocol.
You will have to replace it with the driver on our Github.
I assume you are doing this since you are issue this bug on our repo.
Can you direct me to which dts or dtsi or overlay files you are using?

For your board.

[immagine]<https://user-images.githubusercontent.com/27350014/252625782-3a394cc7-5a5e-4950-abeb-92ce2e7bcfc2.png>

[immagine]<https://user-images.githubusercontent.com/27350014/252626703-037ef33b-75ad-4da8-9478-47e938ef7a9b.png>

[immagine]<https://user-images.githubusercontent.com/27350014/252627073-a14c6627-86da-45c0-8903-db83ad526103.png>

biccius, 2023-07-12:
My first attempt was with the indicated repository version 4.19 but it doesn't boot on my rpi
This is the procedure i've done:

> 
> sudo git clone --depth=1 https://github.com/atmel-maxtouch/maXTouch_linux
> 
> sudo make menuconfig
> 
> Manually enable following items:
> - enable initfsram
> - enable i2c
> - enable touchscreen maxtouch
> - enable modules
> 

.config file generated is available here:
[config.txt](https://github.com/atmel-maxtouch/mxt-app/files/12027093/config.txt)

> Manually edit the dtc-lexer-lex.c file with "sudo nano /scripts/dtc/dtc-lexer-lex.c"
> Find the line 'YYLTYPE yylloc' and change it to 'extern YYLTYPE yylloc'
> 
>
> sudo make -j4 zImage modules dtbs
> 
> sudo make modules_install
> 
> sudo cp arch/arm/boot/zImage /boot/kernel-maxtouch.img

Configure kernel to use new image generate editing "/boot/config.txt"
kernel=kernel-maxtouch.img

After reboot the board freeze and i'm not able to get any information about the missing boot

I honestly have no idea what the problem is.
However, I think that it would be enough for me to have an option in mxt-app that can work with the "-d" option which additionally allows, through a new argument, to specify a possible file containing the progressive used for the sequence number in the calculation of the CRC.

Even better if after reading the infoblock mxt-app was able to understand for example from the family ID if it's an HA module and consequently make the appropriate requests for example create a specific folder for that session.

mgong98, 2023-07-12:
Can you confirm which device tree file you use for your platform.

In our maXTouch Github, I am using a at91-sama5d3_xplained_dm_pda4.dtsi file that describes my hardware resources assigned to the touch controller.

Thanks